bigtreetech/MINI-12864 #3: BTT Mini 12864 incompatible with BTT SKR E3 Turbo

Deebatts, 2021-09-26:
When attempting to compile current Marlin 2.0.9 for my Ender 3 Pro w/ SKR E3 Turbo mainboard, autobuild marlin in VSCode displays an error:
In file included from g:\marlin-2.0.x\marlin\src\pins\pins.h:409,
                 from g:\marlin-2.0.x\marlin\src\inc\marlinconfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
g:\marlin-2.0.x\marlin\src\pins\lpc1769/pins_BTT_SKR_E3_TURBO.h:264:6: error: #error "Only ZONESTAR_LCD, MKS_MINI_12864, ENDER2_STOCKDISPLAY, and CR10_STOCKDISPLAY are currently supported on the BTT_SKR_E3_TURBO."

It seems pretty ass-backwards that your BTT SKR E3 Turbo would not support your own BTT branded 12864 screen; please resolve this conflict. Directions in the user manual seem to be based on an older revision of Marlin also.

radek8, 2021-09-27:
[image]

radek8, 2021-09-27:
[image]

radek8, 2021-09-27:
https://github.com/bigtreetech/BGTHREETECH-MINI-12864-V1.0/blob/main/Hardware/Bigtreetech%20Mini12864%20V1.0%20User%20Manual.pdf

radek8, 2022-06-05:
look here
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650

Obs1234, 2024-02-16:
> look here [bigtreetech/BIGTREETECH-SKR-mini-E3#650](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650)
> 
> After modifying the pin set, configuring Marlin, and making your own patch cord, the display should work.

Will work..but speaker and sd card slot will not work....:(

iliyabathaii, 2025-02-15:
> > look here [bigtreetech/BIGTREETECH-SKR-mini-E3#650](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650)
> > After modifying the pin set, configuring Marlin, and making your own patch cord, the display should work.
> 
> Will work..but speaker and sd card slot will not work....:(

Hello, dear friend, did you solve the problem of sd and but speaker, I also encountered this problem and it was not solved