bigtreetech/MINI-12864 #5: BGTHREETECH-MINI-12864-V1.0 doesn't work with MKS Gen L V2.1 on Marlin 2

romass, 2021-11-11:
Good Day,

I have a problem with connecting this LCD to my MKS Gen L V2.1 on Marlin 2.0.9.1.
LCD lights up, but no visible text appears.
EXP1 wire inverted (if it wasn't nothing would have appeared).
Any help would be appreciated.

Board: MKS Gen L V2.1
Firmware: Marlin 2.0.9.1
LCD: BGTHREETECH-MINI-12864-V1.0

configs: https://github.com/romass/3D-Printing/tree/main/MKS_gen_L_V21_Fysetc%20Mini%2012864%20V2.1%20Marlin%202

MarkPugner79, 2022-02-12:
@romass Good news just did a similar build if you didn't figure it out. The connector body is flipped on the MKS Gen L V2.1 (or on the screen depending on how you want to look at it) I cut notches on my serial connector on the screen so I could plug them in the other way. The red pin should be pin 1 which should be ground. After doing that my screen worked fine after enabling it in the marlin config and setting the board to BOARD_MKS_GEN_L_V21, setting the screen to FYSETC_MINI_12864_2_1 and enabling NEOPIXEL_LED.

romass, 2022-02-12:
Nice! Will try, but I've already swapped one of the connectors 180 degree and at that time it just switched on, but still showed nothing.

radek8, 2022-06-06:
If the display just lights up and shows nothing, try this:
In the file: Marlin\src\lcd\dogm\marlinui_DOGM.cpp
change parameter:
void MarlinUI::_set_contrast() { u8g.setContrast(contrast); }
change to:
void MarlinUI::_set_contrast() { u8g.setContrast(255); }

radek8, 2022-06-06:
@romass 
The setting of your NEOPIXEL parameters does not correspond to the recommendation in the display manual.

[image]