bigtreetech/MINI-12864 #9: Cannot get it to work with an octopus v1.1

thomasrtg, 2022-04-25:
Hi,

I'm a newbie on marlin, I have followed the instructions on the English manual but the screen doesn't show anything, only the LED are On when I turn the motherboard on.
Is the guide un complete or something?
I'll be happy to give you more information if you need.

Thanks in advance

radek8, 2022-06-06:
If the display just lights up and shows nothing, try this:
In the file: Marlin\src\lcd\dogm\marlinui_DOGM.cpp
change parameter:
void MarlinUI::_set_contrast() { u8g.setContrast(contrast); }
change to:
void MarlinUI::_set_contrast() { u8g.setContrast(255); }

radek8, 2022-06-06:
You should also enable NEOPIXEL

[image]